goscommons/CNC-Plasma-Table #10: Torch Height Control

polemidis, 2018-05-04:
The Torch Height Control that is used is the [Proma Electronica SD](http://proma-elektronika.com/index.php/en/products/thc-torch-height-control/compact-thc-sd) and costs $300
I am dumb in electronics, but it should be easy to create a THC for a lot less. 
The process is something along these lines:
The THC get input of the Arc Voltage of the Plasma, divide it 50:1, to a safe level for electronics, apply some filters, and then generate Step and Dir signals according to the voltage read. To high voltage means that the torch is too high, so step/dir signals that lower the Z axis are needed, and the opposite if too low voltage is measured.
If needed we can fast track this by reverse engineering my THC, or some research for other THC available is needed. There are also various plans on youtube, but are beyond my understanding :( I mean it I do not know much about electronics! :)

polemidis, 2018-05-22:
We can collect any THC resources from the net/youtube, and have them all here. So whoever gets involved with that, will have a better picture of what is needed and how its done by others

Witz0, 2018-08-21:
https://www.aliexpress.com/item/Highly-automated-tracking-Portable300-Arc-Voltage-Torch-Height-Controller-THC-Plasma-Height-Controller/32708675367.html

That is an example of less integrated older style electronics likely for a significantly different system. However, the circuit principles are the same. The separate THC is a good design for safety, modularity, and broad compatibility. The Proma THC manual states it has 1:50 divider, which confused me at first until I realized it is a standard option from the arc source and in addition to another HV divider within the THC itself. 

The option from the source is probably better since it probably moves more of the components to the power source and they may be of better quality and give a cleaner signal to start. In that case any voltage divider and filtering on the THC side can be simpler, but this depends on details of that output from the source. Otherwise as the Proma manual states the THC needs HV/HF noise suppression for pilot arc then a HV divider, filtering, an arduino like MC with motor control outputs and optical isolation for safety and signal quality. 

Another way may be to create a HV/HF noise suppression-voltage divider RC combo circuit with an opto-isolated ADC that outputs low voltage values back to the main arduino/RsbPi. This simplifies the circuit of the module that must be attached to the arc power supply. A possible con or issue is the arduino or RsbPi may then need to run more software emualtion like the PID currently on the modular THC to take the voltage inputs from the ADC and modify the Z controller outputs. This seems like a better more integrated solution, but grbl is optimized and likely designed to run on its own on the MC. Unless this can be neatly integrated it may cause problems with grbl. I'm currently uncertain of the RspbPi<-->Arduino I/O and if the RsbPi can assist in any way. A faster arduino version could also be an advanced complex option. 

I'd guess advantages of moving THC complexity back to the main PCB might be more advanced control of THC functions, more accuracy at higher speeds, varying voltage target more accurately based on cornering speed, and by tracking wear on the consumable. But, given the likely software complexity involved keeping a dedicated MC for THC

Overall one of the most difficult parts is THC circuitry safety. It is critical because of the potential for a failure mode where arc power could enter the CNC side and come in contact with human interfaces. Thorough shielding and proper grounding of all PCB's and cables between them is critical to safety.

jurra, 2018-08-21:
From what I get here @Witz0 we are discussing different architecture options for electronics hardware. It would be nice to have a table to compare. It would be great to outline some requirements that sets criterion to make a proper decision.