goscommons/CNC-Plasma-Table #11: Screws have to be added (without threads)

polemidis, 2018-05-04:
All screws have to be added or at least measured in order to be able to provide a BOM

jurra, 2018-05-04:
- [ ] We have to count the screws
- [ ] And classify the different types of screws

polemidis, 2018-05-04:
I propose to replace the Z motor which is a Nema 17, with a Nema 23. The extra torque is not really needed but removes the need for these uncommon M3x45mm. The Nema 23 accepts M5.

jurra, 2018-05-04:
Do you think we are ready to start the BOM, wth the components we have? it would give a rough estimate of the cost right?

polemidis, 2018-05-04:
Yes we could add the more constant expenses leaving the variable ones for the end. 
Can you investigate the market in europe for a plasma cutter that is NOT High Voltage, High Frequency start, but is a blowback or striking system pilot start? I do not know that market. If that helps, in the states a suitable plasma cutter which I have and also saw in videos, is the [Harbor Freight one](https://www.harborfreight.com/240-volt-inverter-plasma-cutter-with-digital-display-62204.html) for an entry level, the [Lotos cut60d](http://www.uwelding.com/supreme-cut60d-60amp-digital-control-cnc-non-touch-pilot-plasma-cutter-with-plasma-gouging-function-non-hf-blowback-arc-start-cnc-plasma-cutter-metal-cutter/) for a better choice which hac already a 50:1 divider fro the THC, and on Hypertherm for the absolute industry workhorse (for double price :)
Anyway, that comment may be as an different issue for BOL

jurra, 2018-05-04:
Yes will take care of that