goscommons/CNC-Plasma-Table #15: Spring tension to restore floating table

polemidis, 2018-05-04:
I created it in the prototype but have to make it in the CAD   ---Pictures needed---

instead of relying on the weight of the floating table+torch weight, a spring has to restore the floating table for reliability and stress free operation

jurra, 2018-05-22:
This is also a simple task right? finding a proper spring that works for the weight of the torch head. @polemidis correct me if I am wrong.

polemidis, 2018-05-23:
yeah pretty super easy. The weight is actually pulling to the same direction as the spring. But we cannot just count on the weight as dirt/friction etc my not be enough force to keep the torch head down. A small spring will do the job. I am satisfied with the rubber band it just look "cheap to the end user :)