goscommons/CNC-Plasma-Table #16: Cable management

polemidis, 2018-05-04:
- [ ] Maybe add to the CAD the Cable chains

- [ ] Design and add to the CAD the support for the Y axis (the long one) cable chain. 

- [ ] Replace in CAD the top "threaded rod plate nema 23" (top plate on Z azis) with a longer one so the drag chain can attach there.

jurra, 2018-05-07:
[image]

jurra, 2018-05-18:
B=50mm width, 18mm height is the ones on the link i believe. Can you open it? The top where you measure the A should be removable so we frst put the wire in, and the close it. It will help 3D printing by a lot since so support is needed.