goscommons/CNC-Plasma-Table #20: New breakout board

polemidis, 2018-05-06:
One bottleneck in the assembly process of the machine is the hude delay in crimping wires so they can will have the the "Dupont connectors that the "CNC Shield"  that I use require. Its annoying and takes huge time.
We have to research for a new board that is compatible with the GRBL v1.1 software, and accepts the wires with no connectors, just simple screwed down.

polemidis, 2018-05-08:
One possible alternative is to replace the CNC shield with a similar board but for Raspberry Pi instead the Arduino. That approach even though it looks more expensive is actually a cheaper option as the Rasp. will replace the computer which runs the grbl sender program. 
[CNC Board for Raspberry Pi](https://blog.protoneer.co.nz/raspberry-pi-cnc-board-hat/)

jurra, 2018-05-08:
Makes a lot of sense! then the user could talk to the raspberry pi in a home network for instance..

polemidis, 2018-05-09:
I have to double check (never had RPI) but if it can run Freecad, and Sheetcam (maybe the best CAM for plasma out there, in reasonable cost) then you just add a monitor/mouse/keyboard and you have your station ready.