goscommons/CNC-Plasma-Table #30: Kit Design of CNC Torch Table

jurra, 2018-07-16:
The general overview of the Kit Concept implies the following design goals:

1. To reduce transport costs, by providing the components, and sub-assemblies as compact as possible
2. To facilitate the assembly of the sub-assemblies and components for the user. Make the life easier for those who want to buy a kit and replicate the machine.
3. Make the tool head easy to exchange to convert the design into a router, a CNC laser cutter, or a Plasma cutter.

polemidis, 2018-07-16:
By saying different approach I actually mean complete redesign. Belts are not good enough to that size for router, so we are actually talking for a new machine. We have to clear this issue before continuing. Router is nice thing to have, but do we need it now? Personally I would love to have a router of that size ;) ;) but I have no time now to research a leadscrew type system

jurra, 2018-07-16:
The backlog items can be prioritized, if this is too complicated, or
unfeasible, we can take it to the bottom of the list, otherwise, maybe some
of us can take over this task.