goscommons/CNC-Plasma-Table #31: Product Backlog Documentation for a Kit design based on the current CNC prototype

jurra, 2018-07-17:
Here we document all the features, improvements and ideas for the new kit design. We are trying out Zenhub to use the scrum workflow. 

The backlog may include sketches, ideas, and specific tasks as well.

JannesHN, 2018-07-24:
My idea of a kit of the CNC plasma cutter is that it only contains (relatively) small essential components.
Parts that we can buy in bulk, such as specific stepper motors, the microprocessor, the small elements in the gantries. The rest, the bigger parts such as the frame beams and rods for the x and y axis should be bought locally by the users.

vinaybhajantri, 2018-07-30:
Hey all, please see the general specifications list based on the small scale CNCs in the market. We should be able to determine these for our CNC too. 

**Note**: I have already put sample values as an example

1. Standard workable area:  3,850 mm
2. Travelling capacity X axis:  3,850 mm without processing of the ends of profile
3. Travelling capacity Y axis:  350 mm
4. Travelling capacity Z axis:  280 mm
5. Fast feed Y - Z:  max. 30 m/min.
6. Fast feed X axis: 50 m/min.
7. Positioning accuracy: ± 0,1 mm
8. Spindle speed: 1-24,000 r.p.m.
9. Power output router spindle: 7,0 kW S1 with constant feedback torque control               
10. Tool support for 5 tools

jurra, 2018-08-21:
For now lets just keep the current specifications we have in the prototype and put them in the readme. Does it makes sense?

jurra, 2018-08-21:
## restructuring product backlog documentation:
I found many comments in separate issues, that can all be gathered here
Maybe this could also go on a seperate markdown file in the root folder of the repository

## Target requirements for CNC table
To be able to have clear cut pieces.
The variables that affect that is
a) Amperage (but for my low end plasma cutter is at 40Amps. It doesnot go more anyway. Ideally for 1/4" we would need 65Amps)
b) Speed
c) Cutting height
d) Piercing height

## Fabrication requirements
The whole machine should be able to be assembled on site.
This means using stockparts, machining and fastening with the leaast amount of different tools.