goscommons/CNC-Plasma-Table #39: Controller: with or without Raspberry Pi

JannesHN, 2018-08-06:
The following discussion took place on the control unit of the machine

**Manolis** 
Did we decide? Arduino + CNC Shield or Raspberry Pi + CNC Hat + Monitor (with the later we can have the software already installed)

**Jannes**
Can both drive 4 steppers? Arduino's main benefit will be the price I guess, while the Raspberry Pi is more sturdy and can be more versatile when we start improving the design or adjusting it for other CNC purposes.
My preference goes out to the latter one, but it makes a huge difference in costs.

**Manolis** 
Let me explain Jannes. The 1st choice is a CNC shield over an arduino. That has the regular (temporary) -non sturdy wire connections, and requires lots of labor intensive clamping of plugs.
The second choice is a CNC Hat that is running the arduino over a raspberry pi. The raspberry pi plays the role of a computer. Its benefits are that we can have the software already installed, it can be a complete solution as a stand alone CNC machine (no need to connect to another computer), and also has regular screw type connectors that save A LOT of time in clamping of delicate wires, and its more reliable/sturdy. The cost is not that much more, but the offer as a complete standalone system can gives as more money as a better product.

JannesHN, 2018-08-06:
Thanks for the explanation, but that does confirm for me that we should go with the Raspberry Pi included.
We want to shake off the DIY image of the machine, so having a more reliable complete product with the addition of a Raspberry Pi seems obvious.

polemidis, 2018-08-09:
I agree. Raspberry Pi with the CNC Hat, is better solution.

closed 2018-08-09 by polemidis