goscommons/CNC-Plasma-Table #40: Packaging design for kit

jurra, 2018-08-06:
## References for packaging design
https://www.ups.com/us/en/services/customized-solutions/optimization-logistics/package-engineering.page

jurra, 2018-08-06:
I had some experience with the packaging of a 3D printer kit. Some of the design of this packaging also applies to our context, specially for electronics, and components. They are the most vulnerable components.

## Question? 
What about the plasma cutter, do we include it in the package or separatedly?

polemidis, 2018-08-09:
I am leaning towards a separate plasma cutter. different grid voltage/frequency accross the globe is hard to follow. I believe each customer has to buy its own plasma, while we can always suggest the models that we know that work in out countries.